rene-dev/stmbl #11: feature request: etherCAT slave

aShure, 2015-11-03:


rene-dev, 2015-11-03:
Hi,
the rj45 sockets are not ethernet, they are used for encoders.
the f405 does not even have a ethernet.
I have looked into ethercat in great detail, but it is very difficult to get the timing right without an ASIC or FPGA. There are a few CPUs which have dual ethernet capability, like the AM3359, which even comes with ethercat examples. But these are not suited for motor control, and that would require a more complex design.
However, we are working on a PCB with a stm32f7 and ethernet, wich can be connected to linuxcnc via a protocol similar to the mesa 7i76E, or used as a standalone motion controller with ethernet.
If you are just looking for a better command interface and using linuxcnc, we are also working on mesa smartserial.
there is also a can port on the IO header, and someone could look into canopen.

Rene

aShure, 2015-11-03:
Hey thanks for the thorough answer!
I know the 8p8c headers are not for ethernet and it sure won't be an easy update. I just love the idea of using a simple ethernet NIC in an x86 machine as the master, without proprietary mesa cards. I thought about something like the LAN9252, never worked with it and I don't know how hard it is to connect to your present stack, but it handles most of the difficult ethercat stuff.
What would be an easy interface to get CANopen out of a LinuxCNC PC? etherCAT -> Beckhoff EK1100 -> EL6751 (that's their CANopen module) would work, but seems bloated.

Sync-, 2015-11-03:
Replacing the Mesa hardware with the LAN9252 replaces one proprietary hardware with another...
With Mesa you at least get the HDL so in theory you could roll your own pretty easily.

Getting CANopen out of a machine is pretty easy, how it would work together with LinuxCNC, no idea.

rene-dev, 2015-11-04:
The mesa cards are not open hardware, but most of their software is GPL and the guy is very helpful.
The LAN9252 looks very interesting! it has SPI, and we have SPI on the extension header, so using a breakout board the stuff could be tested with the current hardware. It does not seem to need much external components. Linuxcnc supports ethercat and canopen, a guy in a german cnc forum uses a EL6751 and Schneider electric drives. I have a EL6751 as well, but have not had any time to play with it.
People say with canopen for motion control is the limited speed.

closed 2016-10-03 by aShure