rene-dev/stmbl #40: pcb version

bronek999, 2016-02-22:
In hw subdirectory is comment v 3.4 but it contain v3.3 files only

rene-dev, 2016-02-22:
V3.4 is in hw/kicad. It will probably be renamed to 4.0 soon.

bronek999, 2016-02-22:
(kicad_pcb (version 4) (host kicad "dummy file") )

it is one line pcb file  stmbl_3.4.kicad_pcb

rene-dev, 2016-02-22:
the pcb is not finished. we are still working on the schematic, and working out pinouts. contributions are most welcome!

crinq, 2016-02-22:
There is a mockup openscad model: https://github.com/rene-dev/stmbl/blob/master/hw/kicad/v3.4/stmbl.scad
I use it to check assembly, isolation distances, air flow, ...
The pcb layout will be based on it.

crinq, 2017-02-20:
files for the 4.1 prototype: https://github.com/rene-dev/stmbl/tree/master/hw/kicad/v4.0

closed 2017-02-20 by crinq