rene-dev/stmbl #43: Supported DC motors?

sjdaliyan, 2016-05-10:
Hi Rene.
I want to know what DC servo's STMBL can support?
If it's possible we talk in email or skype?
sjd.aliyan@gmail.com

crinq, 2016-05-10:
Hi,
the hardware supports DC servos up to 400V and about 20A peak. For feedback encoders (ttl or 1vpp), resolvers and some absolute protocols are supported. Tachos could work too.
The software support for DC is currently broken, but can be fixed in a few days if you can write C and understand our system.
You can contact us over email (renehopf@mac.com or crinq@crinq.de), irc #stmbl on irc.hackint.eu or gitter https://gitter.im/rene-dev/stmbl.

rene-dev, 2016-06-08:
I just added basic dc servo support in the newstuff branch. It works, but all the scaling is off, because it still uses the ac motor model and limits.

sjdaliyan, 2016-06-08:
You mean it runs on sine wave instead of square wave?

rene-dev, 2016-06-09:
no

sjdaliyan, 2016-06-09:
you should put more videos in youtube.
is there anything i can help you with for documentation?

sjdaliyan, 2016-06-09:
I don't find any replacement for SI8621.
Do you know any?

rene-dev, 2016-06-09:
you can use Adum1201, but you have to solder it upside down.
yes, we need documentation, I agree. Do you have hardware?
DC support is now fully working.

Rene

sjdaliyan, 2016-06-09:
no i'm still looking for components.

closed 2016-06-17 by rene-dev