rene-dev/stmbl #53: 24V reset sometimes results in HAL state: CONFIG_LOAD_ERROR

aShure, 2016-09-18:
Power cycling 24V, then trying to connect again in Servoterm:

```
Connecting to /dev/cu.usbmodemFD121
connected
HAL state: CONFIG_LOAD_ERROR
disconnected
```

aShure, 2016-09-18:
Might be because of the psi.txt that I loaded before?

```
psi0.rt_prio = 10
psi0.dc_volt = hv0.dc_volt
psi0.vel = net0.fb_d
psi0.polecount = conf0.polecount
net0.enable = 0
vel1.en = 1
io0.brake = 1
term0.wave0 = psi0.psi
term0.wave1 = psi0.max_psi
term0.gain0 = 500
term0.gain1 = 500

```

crinq, 2016-09-18:
config in flash is only written if you type "save" so psi.txt should not trigger this
sounds like a power startup issue, does your board have the electrolyte cap on the 24V rail?

crinq, 2016-09-18:
psi side note: this config needs a restart of hal to work, type "stop" and "start"
then turn your motor (faster is better) until psi0.max_psi stays constant (the red line), this is your psi

aShure, 2016-09-18:
C34, C48? unpopulated. 

I always did a stop and start after loading psi.txt.

reflashed with `make btburn`, same firmware, newstuff branch. Still 

```
Connecting to /dev/cu.usbmodemFD121
connected
HAL state: CONFIG_LOAD_ERROR
```

but at least no immediate disconnect. I might have altered settings after the psi.txt was loaded and 
then saved. 

list dump shows the psi stuff is still in there:
```
foo0.bar <= foo0.bar = 0.000000
adc0.rt_calc_time <= adc0.rt_calc_time = 0.000000
adc0.frt_calc_time <= adc0.frt_calc_time = 0.000000
adc0.rt_prio <= adc0.rt_prio = -1.000000
adc0.frt_prio <= adc0.frt_prio = -1.000000
adc0.sin <= adc0.sin = 0.000000
adc0.cos <= adc0.cos = 0.000000
adc0.sin3 <= adc0.sin3 = 0.000000
adc0.cos3 <= adc0.cos3 = 0.000000
adc0.quad <= adc0.quad = 0.000000
adc0.res_en <= adc0.res_en = 0.000000
adc0.sin_gain <= conf0.sin_gain = 1.000000
adc0.cos_gain <= conf0.cos_gain = 1.000000
adc0.sin_offset <= conf0.sin_offset = 0.000000
adc0.cos_offset <= conf0.cos_offset = 0.000000
res0.rt_calc_time <= res0.rt_calc_time = 0.000000
res0.frt_calc_time <= res0.frt_calc_time = 0.000000
res0.rt_prio <= res0.rt_prio = -1.000000
res0.frt_prio <= res0.frt_prio = -1.000000
res0.pos <= res0.pos = 0.000000
res0.amp <= res0.amp = 0.000000
res0.quad <= res0.quad = 0.000000
res0.poles <= res0.poles = 1.000000
res0.sin <= res0.sin = 0.000000
res0.cos <= res0.cos = 0.000000
res0.enable <= res0.enable = 0.000000
res0.error <= res0.error = 0.000000
enc_fb0.rt_calc_time <= enc_fb0.rt_calc_time = 0.000000
enc_fb0.frt_calc_time <= enc_fb0.frt_calc_time = 0.000000
enc_fb0.rt_prio <= enc_fb0.rt_prio = -1.000000
enc_fb0.frt_prio <= enc_fb0.frt_prio = -1.000000
enc_fb0.res <= enc_fb0.res = 2048.000000
enc_fb0.ires <= enc_fb0.ires = 1024.000000
enc_fb0.pos <= enc_fb0.pos = 0.000000
enc_fb0.index <= enc_fb0.index = 0.000000
enc_fb0.a <= enc_fb0.a = 0.000000
enc_fb0.b <= enc_fb0.b = 0.000000
enc_fb0.ipos <= enc_fb0.ipos = 0.000000
enc_fb0.sin <= enc_fb0.sin = 0.000000
enc_fb0.cos <= enc_fb0.cos = 0.000000
enc_fb0.quad <= enc_fb0.quad = 0.000000
enc_fb0.oquad <= enc_fb0.oquad = 0.000000
enc_fb0.oquadoff <= enc_fb0.oquadoff = 0.000000
enc_fb0.qdiff <= enc_fb0.qdiff = 0.000000
enc_fb0.error <= enc_fb0.error = 0.000000
enc_fb0.amp <= enc_fb0.amp = 0.000000
enc_fb0.s <= enc_fb0.s = 0.000000
enc_fb0.c <= enc_fb0.c = 0.000000
encm0.rt_calc_time <= encm0.rt_calc_time = 0.000000
encm0.frt_calc_time <= encm0.frt_calc_time = 0.000000
encm0.rt_prio <= encm0.rt_prio = -1.000000
encm0.frt_prio <= encm0.frt_prio = -1.000000
encm0.pos <= encm0.pos = 0.000000
[... 567 lines cut ...]
conf0.sin_offset <= conf0.sin_offset = 0.000000
conf0.cos_offset <= conf0.cos_offset = 0.000000
conf0.sin_gain <= conf0.sin_gain = 1.000000
conf0.cos_gain <= conf0.cos_gain = 1.000000
conf0.max_dc_volt <= conf0.max_dc_volt = 370.000000
conf0.max_hv_temp <= conf0.max_hv_temp = 90.000000
conf0.max_core_temp <= conf0.max_core_temp = 55.000000
conf0.max_pos_error <= conf0.max_pos_error = 1.570796
conf0.high_dc_volt <= conf0.high_dc_volt = 350.000000
conf0.low_dc_volt <= conf0.low_dc_volt = 12.000000
conf0.high_hv_temp <= conf0.high_hv_temp = 70.000000
conf0.fan_hv_temp <= conf0.fan_hv_temp = 60.000000
conf0.fan_core_temp <= conf0.fan_core_temp = 450.000000
conf0.fan_motor_temp <= conf0.fan_motor_temp = 60.000000
conf0.p <= conf0.p = 0.990000
conf0.pos_p <= conf0.pos_p = 100.000000
conf0.vel_p <= conf0.vel_p = 1.000000
conf0.acc_p <= conf0.acc_p = 0.300000
conf0.acc_pi <= conf0.acc_pi = 50.000000
conf0.cur_p <= conf0.cur_p = 0.000000
conf0.cur_i <= conf0.cur_i = 0.000000
conf0.cur_ff <= conf0.cur_ff = 1.000000
conf0.cur_ind <= conf0.cur_ind = 0.000000
conf0.max_sat <= conf0.max_sat = 0.200000
```

crinq, 2016-09-18:
try mounting c48, c34 xor c70
value is not critical

aShure, 2016-09-18:
Power cycled. Magically working again: 

Connecting to /dev/cu.usbmodemFD121
connected
conf0.j = 0.000026

aShure, 2016-09-18:
Btw. 24V in rush current is limited to 200mA at the lab power supply to not fry anything accidentally. HV side is connected to a second PSU (off right now)

crinq, 2016-09-18:
CONFIG_LOAD_ERROR can also happen if you switch branches, as we sometimes change config stuff
if so write a new config with "save" to fix this

aShure, 2016-09-18:
ok good to know. 
Is it normal that the F4 is quite hot? The board pulls 140mA with a 5V encoder, 90mA without. The 7812 is close to frying pan mode, might change it to a small DCDC module.

crinq, 2016-09-18:
7812 should have normal temp if no fan or 12V FB is connected
f4 reaches 50-60°C at normal ambiente temperature

aShure, 2016-09-18:
Ok good to know. 7812 is slightly too hot to touch.

crinq, 2016-09-18:
side note: there is no esd protection on the io and swd headers on stmbl version <= 3.3
esd can fry gpio pins, wich results in higher current consumption
140mA - 160mA is normal (depending on the batch and used periphery)

aShure, 2016-09-18:
Power cycling results in a fully working state if the encoder is not connected when the 24V is enabled. It's fine to connect it after the F4 is awake. A slow rise time for 24V might be a problem for the F4.

crinq, 2016-09-18:
try with a higher current limit and/or some caps on the 24v rail
in v4.0 FB voltage is disabled on startup

rene-dev, 2016-09-18:
CONFIG_LOAD_ERROR means the config flash section could not be read, or has an incorrect format. This happens occasionally, but I have no idea why, and what actually happens. Could be debugged by giving more details in the error message.

rene-dev, 2016-09-18:
Try typing load when this happens. It might already give you a more detailed error message

rene-dev, 2016-10-01:
EE_ReadVariable returns 1: the variable was not found
on reading address 0
https://github.com/rene-dev/stmbl/blob/master/src/eeprom.c#L266

closed 2016-11-26 by aShure