rene-dev/stmbl #58: How to deal with different encoder type with the same protocol?

sacidu93, 2017-02-05:
I made a custom hardware based on the stmbl V4, and I managed it  to work with Yaskawa servomotor. As my servomotor use 13bit incremental encoder instead of 16bits absolute encoder, the received frame data from this encoder is different, and contain more interesting data field like the Hal sensors state, compensation data and status bit that indicate if this last field is valid or no. 

position 7: one bit indicate if Z index has been observed at least once since powering up the encoder, this is used to indicate if the compensation data is valid, ( 1: data not valid, 0: data is valid ) the absolute shaft angle can be calculated by subtract the compensation data from the incremental position data counter.  

Position 58, 59 and 60 are the Hal sensors state ( U, V, and W ), can take six value (2, 3,1, 5, 4 and 6 ) this sequence repeat 3 times per revolution ( All Sigma II SGMAH series are 3 pole-pairs motors ) this signal can be used to do initial phasing without enabling auto-phasing feature . 

Position 67:  13 bits of incremental position data.

position 81:  11 bits of compensation data ( need to be 2bits left shifted if used as absolute offset from Z index)

My question is how to deal with this kind of situation, 16bits absolute, 17bits incremental, 13bit incremental or absolute, 20bit and higher for Sigma III and Sigma V.. and the same situation with Mitsubishi encoders.

Is it doable to make each  encoder type in separate .COMP file and rename it some thing like:

"Yaskawa_13b_inc.comp", "Mitsubishi_17b_abs.comp"...

or may be use only one .comps file that can handle all types but in this case each protocol must have sub-non-volatile parameter to indicate how to decode data frame.

Here a .comp file that handle Yaskawa Sigma II 13bits incremental  encoder ( SGMAH series ) using FB0_45  ( some optimization still to be done ) and the config file for SGMAH-04-AAA .

[yaskawa_13b_inc_FB0_45.txt](https://github.com/rene-dev/stmbl/files/752872/yaskawa_13b_inc_FB0_45.txt)
[Yaskawa_SGMAH-04AAA.txt](https://github.com/rene-dev/stmbl/files/752873/Yaskawa_SGMAH-04AAA.txt)

crinq, 2017-02-20:
We don't have a good solution for protocol variants or hw instances yet. There are multiple options:

1. multiple comps (one / variant, see enc_fb.comp and enc_cmd.comp)
 "+" fast
 "-" code duplication

2. preprocessor switch in comp
 "+" fast
 "-" ugly code
 "-" config at compile time

3. runtime switch in comp
 "+" config at runtime
 "-" slow

4. load time switch in comp (bend function pointers in RT_INIT)
 "+" fast
 "-" needs a lot "setup" code (could be implemented in HAL)

5. multiple comps (hw, protocol, interpretation)
 "+" separation of concerns
 "-" runtime overhead
 "-" communication is limited to float or float[]
 "-" complex HAL graph

Currently i would prefer option 4 with better support in HAL.

andypugh, 2017-03-30:
The Mesa BiSS / SSI etc functions handle this using a string parameter that encodes the number of bits and their HAL pin names:
http://linuxcnc.org/docs/2.7/html/man/man9/hostmot2.9.html#Synchronous Serial Interface (SSI)
This might have too much overhead for STMBL, however. It re-uses the code that allows smart-serial devices to choose their own pin-names and types.

rene-dev, 2017-03-30:
good idea, but our hal implementation cannot do that.
components and their pins are fixed at compiletime.
I also support option 4.

andypugh, 2017-03-30:
Components and pins _used_ to be fixed at compile-time in LinuxCNC too. I had to change that to support smart-serial. It's just a matter of keeping the pins in a Union with a type code:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/drivers/mesa-hostmot2/sserial.c#L1385
But I did mention that I thought there might be too much overhead for STMBL.

corrylaz, 2019-06-13:
This isn't related to the issue, but sacidu93, do you have information on the sigma-V protocols?  The project I wasn't to use the STMBL on I need to read some existing Yaskawa sigma-V motors for position information.  Can you point me to information on reading the encoder data?

HUANGJIN169, 2020-11-27:
Hello, first of all thank you for your project
Does this project support the 22-bit encoder in the picture?
[image]

https://www.google.com/imgres?imgurl=http%3A%2F%2Fimg30.360buyimg.com%2Fimgzone%2Fjfs%2Ft1%2F58475%2F25%2F2185%2F118223%2F5d021e66E6c388d4e%2F2822882d565505a9.jpg&imgrefurl=https%3A%2F%2Fi-item.jd.com%2F50031585455.html&tbnid=DC6OU_J7Sk47SM&vet=12ahUKEwiDk6376KHtAhUIZ5QKHUHuDJEQMygKegUIARCmAQ..i&docid=HXahkNw1rqnPeM&w=750&h=460&q=%E9%95%BF%E6%98%A522%E4%BD%8D%E7%BC%96%E7%A0%81%E5%99%A8&ved=2ahUKEwiDk6376KHtAhUIZ5QKHUHuDJEQMygKegUIARCmAQ

crinq, 2020-11-27:
@HUANGJIN169 that's an incremental encoder with 2500 pulses per revolution

HUANGJIN169, 2020-11-27:
Ok thanks for your reply

HUANGJIN169, 2020-11-27:
Hello there
Can this driver board flash other firmware?
For example, MMOS ffb simulation racing force feedback firmware

HUANGJIN169, 2020-11-27:
I want to use this drive board to imitate the analog racing control drive
board
The following is my project and MMOS ffb force feedback simulation racing
car:
https://hackaday.io/project/168801-diy-mmos-ffb-stm32-and-servo-drive-servo-motor
MMOS firmware: https://forum.virtualracing.org/showthread.php/92420

crinq, 2020-11-27:
Yes you could flash other firmware, but you have to modify it. MMOS ffb will not run out of the box.
You can use STMBL to drive the motor and run stock MMOS ffb on a different board.

HUANGJIN169, 2020-11-27:
MMOS ffb will not be used immediately
I know this
Need STM32 ST-LINK utility tool to refresh MMOS firmware
I know

HUANGJIN169, 2020-11-27:
Hello big brother
do you have Facebook?
Or discord and other ways to contact you
PCB file printed circuit board in file
The print shop says my files are not displayed
So I still have questions and need your help
First thank you

artayasa, 2024-01-11:
> good idea, but our hal implementation cannot do that. components and their pins are fixed at compiletime. I also support option 4.

Hi sir it posible to modif the hardware sir? I want to used only main board with stm32f4, i just want to used arount 40-60v dc input sir. Or that need modif the frimeware also?