rene-dev/stmbl #64: Issue with kicad

CarrelN, 2017-06-13:
The version 4.0.6 of kicad (last one) can't open files which you gave in kicad folder. The error said that we need a recent version of kicad compatible with the version 2016 but we have the last version possible. Can you post the BOM and the gerber files to produce some boards for version 4.0 ?
This is my mail for contact: carrel.nirina@gmail.com

Thank you.

rene-dev, 2017-06-13:
use kicad nightly.
https://github.com/rene-dev/stmbl/issues/62

CarrelN, 2017-06-14:
Hello,
I've the same error also with the nightly build when i tried to open STMBL
V4 with Kicad. You can find in attachement the error that I had. And also I
use the version kicad-r8167.e97f49e00-x86_64.exe
<http://downloads.kicad-pcb.org/windows/nightly/kicad-r8167.e97f49e00-x86_64.exe>
of
kicad nightly.

Kind regards,
[image: Images intégrées 1]

closed 2017-11-28 by rene-dev