rene-dev/stmbl #91: IRAM256-2067A2 Obsolete and out of stock

maxschommer, 2018-05-06:
IRAM256-2067A2 is obsolete and out of stock everywhere I looked.

crinq, 2018-05-06:
Yes, Infineon discontinued it after they bought IR.
New design is in the pipeline.

rene-dev, 2018-05-06:
IRAM256-1567A2 or IRAM256-1067A2 can be used as replacement, but they only have 15 and 10A, but thats plenty for most servos.
they are still available at mouser.

stupotmcdoodlepip, 2018-05-24:
Have you chosen a new part for the next revision yet? I'd like to get some boards made soon so would like to start gathering components. 
Many thanks for this project!

rene-dev, 2018-05-24:
the next revision is far from complete, dont bother stocking parts. Make v4 if you cant wait.

rene-dev, 2018-06-21:
All A2 models seem to be out of stock now. The A models can be bodged in:
[image]

buleks, 2018-09-03:
Any progress with new revision?

crinq, 2018-09-03:
yes, but not ready yet

meet-gopani, 2018-10-12:
both IRAM256-1567A2 and IRAM256-1067A2 are also discontinued by infineon, can you help guide selecting a similar driver.

meet-gopani, 2018-10-13:
can we replace IRAM256-2067A2 with ON semiconductors- STK551U3A2A-E ?
http://www.farnell.com/datasheets/2354470.pdf?_ga=2.109968454.293445377.1539426500-505894212.1539426500

crinq, 2018-10-13:
No this does not work with 3 low side shunts (connected emitters of the lower igbts).
The next STMBL version has a discrete 3 PH bridge with TO220, but the design is not ready yet.

meet-gopani, 2018-10-13:
okay, thank you!
STK554U392A-E: this is 15A version and this works with 3 low side shunts
http://www.farnell.com/datasheets/2354469.pdf?_ga=2.30356220.293445377.1539426500-505894212.1539426500
will it work?

corrylaz, 2019-01-03:
I know I feel like a total nag.  I was going to build one of these, the bought a full CNC instead....but it turns out that machine can't get "Polygonal turning" (I want it for gear hobbing) so I'm going to add a spindle I can synchronize.  I was planning on the full 2Kw.  I may see if I can beg for one of the older chips.  I'm not requiring this....now....as I've got some other design and building to do of the mechanical parts.  Any hints as to when it will be out?  Any beta's?  as long as its not going to catch on fire, I'm game :)

acetylen-joe, 2019-02-19:
I have a source, they got 20 units of the IRAM256-2067A2  in Stock left, but these costs 44,06€ per piece. If you think it is worth it i can order these IBGT Modules. I am from Germany (Berlin).

awigen, 2019-02-19:
I'd like to order three!

corrylaz, 2019-02-20:
I’m no github expert, otherwise it contact you off of this (if it’s possible).  I’d order 4.  If you can get my contact info, just contact me direct.  I’m in the USA, so I know shipping won’t be the easiest!

silopolis, 2019-02-20:
Hi,

Le mar. 19 févr. 2019 à 14:16, acetylen-joe <notifications@github.com> a écrit :
>
> I have a source, they got 20 units of the IRAM256-2067A2 in Stock left, but these costs 44,06€ per piece. If you think it is worth it i can order these IBGT Modules. I am from Germany (Berlin).
>

Currently in planning/design/sourcing stage of a metal mill build that
I want to be as Open Source as possible, I stumbled this week on this
project and would love to be part of a group buy as I sadly discovered
yesterday that supplies were an issue until next version (not before
months).
I'd need at least 3 drives to begin with for axes, may be a 4th one if
I can drive my spindle motor with STMBL. Next, there should be one
more for the ATC, and two others for the rotary axes, but a handfull
of versions should have been released by that time ;) :D
Would be even better if part of my money could benefit the project and
help further development.
And I'm in south of France FWIW.

rene-dev, 2019-02-20:
sounds interesting, any more info about the machine?

acetylen-joe, 2019-02-20:
@rene-dev can yo give a guess or an etimation when the next version / generation of the STMBL Board will be published - that means without the  obsolete IRAM Chips from Infineon therefore with an other IBGT module? 

I am undecided or it is worth to build the actual design with the last IRAM256 Chips if maybe in a couple of weeks there will be no more need for this IBGT Module and it is replaed with a new Version with estimated Production Lifetife for ne next 10+ years. And of course a lot cheaper than the last IRAM Chips in Stock now.

Sync-, 2019-02-20:
It won't be ready in the next couple of weeks.

acetylen-joe, 2019-02-20:
so we are talking about months or years?

silopolis, 2019-02-20:
Hello Rene,

Le mer. 20 févr. 2019 à 11:30, Rene Hopf <notifications@github.com> a
écrit :

> sounds interesting, any more info about the machine?
>
Don't want to hijack this thread so, shortly, project is a 500x250x250 mm
work enveloppe metal mill (Z height provision for trunion for future
upgrade to 5 axes), ball bearing rails, ballscrews, servo driven axes,
linear encoders feedback for positioning, 3HP and above servo driven
spindle (ridgid taping etc), xx30/40 tool holding, automatic tool change,
auto lubrication, flood and mist coolant (hopefully center of tool too),
controled by LinuxCNC (or MachineKit ?). Should maximize use of standard
compliant COTS parts and design should minimize (if not avoid) need for
large machining, treatment operations. Structure will be reinforced cast
epoxy concrete (alternatively high strength concrete) or epoxy concrete
filled steel tubes/profiles.
All documentation, design models, plans, BoMs, etc. will be Open Sourced
(and published here on GH), LinuxCNC is of course FOSS, I now have found
OSHW servo drives (yeah !), Mesa cards at least have their firmware Open
Sourced so, that's not so bad :) You guys know of OSHW BLDCs ?! Everything
else should be field replaceable by widely available (non propietary !)
parts.
Machine should be built minimal with 3 axes and produce (most of) its own
upgrades (mainly ATC, 4th axis, 5th axis trunion,...).
Well, here we are, that wasn't short :P

5tran9er, 2019-02-20:
Hi guys.
Who is interested in IRAMs can find some at tme.eu
At the moment they declare 63 pcs in stock.
Thee ones that I ordered came from Poland as I remember
The price is 22€ each for 3+ pcs

beelash, 2019-02-20:
This is a different version. iram256-2067a2 is needed

acetylen-joe, 2019-02-25:
If the Vers. 4.1  of the stmbl works with the IRAM256-2067A Module, i will give it a try. But i have one Question left: whats the difference between the IRAM256-2067A and the IRAM256-2067A2 IGBTs?

rene-dev, 2019-02-26:
the way the pins are bend.
https://github.com/rene-dev/stmbl/issues/91#issuecomment-399050388

awigen, 2019-02-26:
@rene-dev STK554U392A-E looks to be pin compatible, do you have any thoughts on this?

https://uk.farnell.com/on-semiconductor/stk554u392a-e/motor-driver-3-phase-ac-15a-sip/dp/2424504

silopolis, 2019-03-01:
Aren't these the good ones ?:
https://www.electronics-distributor.com/components/International-Rectifier(Infineon-Technologies)/IRAM256-2067A2.html
https://www.avnet.com/shop/us/products/infineon/iram256-2067a2-3074457345624908256/
https://www.digipart.com/part/iram256-2067a2?gclid=Cj0KCQiAzePjBRCRARIsAGkrSm5PHYnI2n_eZbIYmGF9fHAkB9kmrR6MpJyiG4In5JWSB0iJVP3CEbwaAi1gEALw_wcB

fehrudi, 2019-03-08:
Infineon IKCM series seems to be a good replacement, for example IKCM30F60GA. It would require only minor modification to the existing design.

kamilkamil1, 2019-03-21:
Hi Rene,
I am very impressed with you project. I know the question would I ask was asked many time but answer was  not exactly for my question. Could you tell me if the new version of stmbl  for TO220 will change only the HV power board or also control board  ?

Thank you.

crinq, 2019-03-21:
the logic board will change slightly to keep the isolation distance (the cutout for the HV mounting screw moves a bit)

jagdeepsingh02, 2019-09-15:
Hello @crinq any news on the new version? thanks :)

asorien, 2020-01-05:
Hi ! I have a question before starting to build the v4.1 stmbl : all versions have the same pcb layout ?

asorien, 2020-01-05:
from were I can download the pcb layout and who make it the cheapest ? Thank s!

crinq, 2020-01-17:
V4.3 pcb data is there: /hw/kicad/v4.0
but you should try to source the iram first

kamilkamil1, 2020-02-03:
Hi Crinq, 

Can I omit some components (not solder, eg SN65176B) if I need only encoder and step/dir inputs ? I want to check firstly the connection between PC and STMBL board. 

Thank you.

rene-dev, 2020-02-03:
yes, that is possible.

kamilkamil1, 2020-02-04:
Thank you Rene,

STEPPING3DCAD, 2020-02-12:
Shouldn't this new infineon IM393X6FXKLA1 part work??  It looks like the replacement for it, although they don't say that... It's obviously not directly pin-compatible, but the data sheets are almost identical.

https://au.mouser.com/ProductDetail/Infineon-Technologies/IM393X6FXKLA1?qs=sGAEpiMZZMtKB4wrjsn3lUUtnZ7xtd6RKKFrk9F5T%2Fg%3D

crinq, 2020-02-12:
DIP parts don't work in the current form factor.

ShadeTechnik, 2020-03-15:
The IM393X6FXKLA1 is a SIP part but it doesn't have the forward bent legs like the A2 version of the original power IC. It would be a similar bodge to the "A" model as for angle, but the pinout is different. With re-routing of the power board it looks like it would work but the angle is not good for the form factor

Reworking the power board -for- the DIP package would actually probably be the easiest scenario as the power IC can still sit on the bottom of the PCB screwed against the heatsink. I may actually give this a try, the power PCB section is alot more simple to deal with, it's mostly just mechanical concerns with mounting. I didn't fully study the datasheet, but I think the IKCM15L60GDXKMA1 would work for this.

rene-dev, 2020-03-16:
there is no space to re route it properly with this formfactor, feel free to try.

pacoespinola1, 2020-06-04:
is there somebody trying a solution?
I am going to try may be the IM393X6FXKLA1.

krzysiek-007, 2020-06-07:
Hi.
Anyone can sell me 1 pcs stmbl?
Sorry for my English

pallavi339, 2020-07-18:
IRAM256-2067A2 is out stock everywhere

metacomgd, 2020-12-09:
Rochester electronics have ~4k in stock, MOQ of 17 at $15.28 a piece
https://www.rocelec.com/part/INFIRAM256-2067A2

elektonik-computer, 2021-12-29:
Hello, did I understand correctly the IRAM256-2067A2 can I replace with the IRAM256-2067A in the PCB version STMBL 4.3?
Thanks Axel

crinq, 2021-12-29:
yes, but you need a custom heatsink/case/mounting solution

fettah2, 2023-08-20:
Hi,  it may be a bit late, but please ShadeTechnik have you succeeded with  IKCM20L60GDXKMA1  or other part?
any news for this wonderful project ??

artayasa, 2023-11-20:
> All A2 models seem to be out of stock now. The A models can be bodged in: [image]

Hi sir it posible to change the iram with this chip IKCM30F60GA??? 
And also i just want to used step dir comunication only ( i want to used mach3) so can i remove some parts??? To save more money??.