rene-dev/stmbl #93: Capacitor voltage rating

professor-fungus, 2018-06-01:
I apologize if I have overlooked something.

Reading through the BOM, I cannot find voltage specifications for capacitors. This is not labeled on the schematic either from what I can tell (in KiCAD).

Are the majority of the capacitors a set voltage? If so, what voltage specification should I work toward?

Thank you for the development of this amazing project. I look forward to implementing this in many applications.

rene-dev, 2018-06-01:
there are a few BOMs with part numbers:
https://github.com/rene-dev/stmbl/wiki/PCB

professor-fungus, 2018-06-01:
The BOM for v.3.3 is very comprehensive, listing part numbers and full specifications; however, I can only assume that the reference designators (as well as the architecture) has changed between v.3.3 & v.4.1. Again, I may be overlooking something. Here is an example from v.4.1:
[image]

crinq, 2018-06-01:
v3 and v4 are completely different schematics, but they share most of the passive parts. 
2.2µ, 100n, 1n, 18p > 30V, 10µ > 16V, 22µ > 24V, 270µ >= 400V

professor-fungus, 2018-06-07:
Perfect! Thank you for the values. If I create an arrow.com cart, would you like me to share it?

Thanks again for the help.

closed 2018-06-07 by professor-fungus